team-rocos/platform #3: Need improved workflow for connecting robots.

edwinhayes, 2019-02-05:
The workflow for connecting a robot to ROCOS seems only suitable for connecting individual robots in a test environment.

1. `rocos-agent --init` is interactive; it could easily take comment line arguments instead.

2. The authorisation process is basically impossible to automate, impossible run on a computer which isn't being accessed by SSH but doesn't have a GUI, and just seems backwards.  We would rather enter some static information ahead of time on the robot side, then have an administrator approve one or more incoming connection from robots via the web interface.

rstinear, 2019-02-06:
Thanks for the feedback @edwinhayes - we had some concepts for this and for #4 on our roadmap so good to see them reinforced - I've created #6 and #7 to roll up your suggestions and track them as enhancements.

rocos-david, 2019-03-04:
Will be tracked from #6 and #7

closed 2019-03-04 by rocos-david