team-rocos/platform #5: Agent - optional installation as service

rstinear, 2019-02-06:
During the agent installation process, installing the agent as a service should be optional for the user (raised from #4)

1. Take this as a flag on the command line during init
2. Provide it as an option in the wizard flow

edwinhayes, 2019-02-06:
Just to clarify, from my point of view, creating the service unit file is fine, just not enabling it.  In our case,  we'll probably override it with a `/etc/systemd/system` version that includes the original but adds additional deps and configuration such as syslog stuff.

rstinear, 2019-02-06:
Yes that sounds totally sensible - we'll have the service config ready to go but will only actually enable it with an explicit opt-in from the user.