team-rocos/platform #8: Agent - Sensible reentrant behaviour.

edwinhayes, 2019-02-06:
The agent binary is kinda weird, because it's both the setup wizard and the actual daemon; the first time you run it, it exits; the second time, it's blocking.  And if you run the binary while the service is running, then the behaviour is not sensible: it runs as a duplicate instance, then segfaults in the MAVLink plugin.

Sensible behaviour would be something like:

a. The first time you run `rocos-agent` you get the wizard.
b. Any subsequent time you run `rocos-agent --init` you get the wizard.
c. Any subsequent time you run `rocos-agent` it just gives you a status message regarding the state of the agent daemon, but doesn't start/stop anything.
d. To run the agent as the daemon, the service should call `rocos-agent -d`.
e. Calling `rocos-agent -d` while an agent is already running should error out.

rstinear, 2019-02-06:
Thanks - I've marked this one as a defect as step (c) should indeed tell the user the current status if it's currently being run in the background. We'll tidy up the flags too for daemon mode - this will give the user more options for how they wish to run the agent in the background.

rstinear, 2019-04-08:
Fixed in release [v0.0.5-3163](https://docs.rocos.io/beta/docs/agent-release-notes)

closed 2019-04-08 by rstinear