team-rocos/platform #10: Portal not loading on Safari 12

rstinear, 2019-02-07:
**Describe the bug**
Raised through #9. Safari 12 display a syntax error in its console: [Error] SyntaxError: Invalid regular expression: invalid group specifier name
(anonymous function) (main.89877e72b4300c7b0558.js:1)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to portal.rocos.io in Safari 12
2. Observe blank page

**Expected behavior**
Portal should load

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

e10101, 2019-02-08:
It is a regular expression bug on Safari browser. Bug fixed! This fix will be released in the next version.

rocos-yishi, 2019-02-08:
It is a Regex bug on Safari browser. 

The bug has been fixed and it will be released in the next version.

edwinhayes, 2019-02-08:
Can you confirm what "released in the next version" means in terms of updates to the agent/backend/portal during this closed beta?

rstinear, 2019-02-08:
@edwinhayes we decided to do it as a hotfix instead, so the portal has been updated just to resolve the regex issue.

In terms of future releases - we'll provide some information closer to the next release in terms of new features, any impact to users etc.

closed 2019-02-08 by rstinear

r3mafe, 2019-03-14:
> It is a Regex bug on Safari browser.

Could you please provide a proof link? Facing similar behaviour on another software piece and can't find any relative information.

rocos-yishi, 2019-03-14:
@r3mafe 
Have a look: https://stackoverflow.com/a/51568859/4288118