team-rocos/platform #9: Portal status

edwinhayes, 2019-02-07:
portal.rocos.io has been a blank page since yesterday; I assume you're probably hard at work on something, but I can't tell for sure.  You might need some kind of status.rocos.io page so users know what is going on?  Or update twitter or something.

rstinear, 2019-02-07:
That's an odd one! It's been working fine for us. Would you mind letting me know:

1. Which browser you're using
2. Whether there is any error output in the console of the dev tools for whichever browser you're using
3. Are you getting a 404 or just a blank page?

edwinhayes, 2019-02-07:
Safari 12.  Ah, probably this:

[Error] SyntaxError: Invalid regular expression: invalid group specifier name
	(anonymous function) (main.89877e72b4300c7b0558.js:1)

Just blank.

rstinear, 2019-02-07:
Ah good catch - looks like a Safari compatibility issue. Will feed it into the team.

edwinhayes, 2019-02-07:
Ok cool.  I guess my original point remains: there needs to be some way to confirm what the status of the service is; also would be helpful on the agent side - I'm not really sure how to tell from the agent side whether it is connected to your backend, which might be helpful if trying to diagnose something in the field.

rstinear, 2019-02-08:
Great suggestions! I've added #11 and #12 to cover this one off.

closed 2019-04-02 by rocos-david