team-rocos/platform #16: Agent installs non-executable files into 'bin'.

edwinhayes, 2019-02-24:
At the moment, all the agent's configuration files are being installed into `/opt/rocos-agent/bin`.  They aren't binaries, they should be going into `/opt/rocos-agent` or some other subdirectory instead.

rstinear, 2019-02-25:
Thanks, the plan is to move them into somewhere under /etc, and we'll address this when we clean up the rest of the agent installation and run issues.

edwinhayes, 2019-02-25:
Ok cool.  Also, it looks like you create a dedicated user for rocos-agent; the user definition specifies a home directory at `/home/rocos-agent` which doesn't exist, and leaves the shell entry blank instead of _nologin_.

rstinear, 2019-04-08:
Fixed in release [v0.0.5-3163](https://docs.rocos.io/beta/docs/agent-release-notes)

closed 2019-04-08 by rstinear