team-rocos/platform #18: Handle multi-component MAVLink systems correctly.

edwinhayes, 2019-02-25:
A MAVLink namespace can consist of multiple 'systems' (where broadly, one 'system' is one mobile robot), which each consist of multiple 'components' (individual nodes which talk via MAVLink, both physical and logical).

At the moment, the pipeline for MAVLink messages (can't tell if it's on the agent side or the server side) does not handle systems with multiple components correctly.

This is probably not noticeable if you are just testing against simple drones, since most low end aircraft out there only expose a single MAVLink component (the flight controller).  But more complex aircraft might have a dozen different components.

Right now, I see heartbeat MAVLink messages appearing in the data explorer, and nothing else.  Based on the mode reported in the heartbeat messages, it looks like it's the heartbeat from a node which _only_ sends heartbeats.  So I think you're probably just selecting the first component ID which you see, and then ignoring messages from all others?

Also, a related thing: because of the fact you can have multiple components, your data model needs to expose the senders component ID for each MAVLink message.  System ID is probably not required, since it's reasonable to assume that you have one instance of the agent per system.

rstinear, 2019-02-27:
Thanks @edwinhayes, we are making some changes to the plugin design so we can instantiate multiple plugins of the same type which should take care of this, i.e. the config for each instance of the plugin will contain desired component ID, port and so forth. Again - great feedback thanks!

rstinear, 2019-02-27:
Thinking about this a bit more, amd just to double-check - is that heartbeat message you're seeing definitely a MAVLink heartbeat? e.g.

```
{
  "CUSTOM_MODE": 0,
  "TYPE": 2,
  "AUTOPILOT": 3,
  "BASE_MODE": 81,
  "SYSTEM_STATUS": 3,
  "MAVLINK_VERSION": 3
}
```

edwinhayes, 2019-02-27:
Yup.

```
{
  "CUSTOM_MODE": 0,
  "TYPE": 18,
  "AUTOPILOT": 8,
  "BASE_MODE": 192,
  "SYSTEM_STATUS": 4,
  "MAVLINK_VERSION": 3
}
```

Note that `type` field: 18 is `MAV_TYPE_ONBOARD_CONTROLLER` (can be looked in up the common.xml MAVLink defs, available at mavlink.io).  So this is something running on our companion computer.  The actual flight controller would be a different type: `MAV_TYPE_QUADROTOR`.

For me, I see that one message statically displayed in the data explorer; what I'd expect if for that message to be rapidly changing, since there are actually about a dozen heartbeat messages every second.

edwinhayes, 2019-02-27:
I would consider it be sane to expose this as `/mavlink/s1c1/heartbeat`.  Or as `/mavlink/c1/heartbeat` although in that case you also need to consider that while your abstraction probably expects each system to have one agent, and thus be one robot, mavlink devices may still be getting messages from other systems (for instance, the ground station calling itself S255 is pretty standard).