team-rocos/platform #19: Can't make a graph of CPU usage.

edwinhayes, 2019-02-28:
It's not possible to make a line graph showing CPU usage, because (on a multi-core machine at least), the JSON encapsulation of the CPU info isn't in the expected key:value format.

neno-k, 2019-02-28:
hi @edwinhayes , yup we are aware for this one and working on a solution

edwinhayes, 2019-03-01:
Hmm, I note this is also the case for data coming in via rosbridge, which seems weirder.

rocos-david, 2019-04-02:
Current work around to view this data is to use the new Teleop "Table" Widget on a dashboard, this can show the numeric values as text rather than a graph.

edwinhayes, 2019-04-02:
Well, what's weird now is that `/system/cpu` now only has one element for me, even though it's a quad-core machine the agent is running on, and there used to be all four cores displayed.

edwinhayes, 2019-04-02:
Oh, hang on, it's not just CPU: `/system/network` is now only showing one interface (looking at the JSON via the data viewer page), which is also very incorrect.

edwinhayes, 2019-04-02:
I'll raise a separate issue.