team-rocos/platform #20: Storage streams.  Do they work?

edwinhayes, 2019-02-28:
Just to check, do you expect 'storage streams' to actually work right now?  They don't seem to do anything for me.  Or is this another Safari related issue?

Shown below is what a configured stream (to a _real_ robot) looks like for me.  It doesn't change, the graphs are basically static, which isn't what I'd expect for something which is streaming CPU data at 1s intervals.

[image]

Also, where is this storage data going?  How can I access it ever again?

neno-k, 2019-02-28:
Hi @edwinhayes, I've passed this to the team to investigate, believe there is an issue with the graph rendering the /system/network and /system/cpu metrics due to the way the payload is formatted, I will let you know once this is fixed

neno-k, 2019-02-28:
@edwinhayes , can you also please confirm that you are not having the same issues with other stream definitions eg /mavlink/vfrhud

neno-k, 2019-02-28:
@edwinhayes quick update on this, team has replicated this and it looks like an issue with safari, I will give you another update by the end of the day - thanks

edwinhayes, 2019-02-28:
Graphs have started appearing, so looks like you've fixed something!

edwinhayes, 2019-03-03:
I note that the requests/second graphs have some weird artefacts at the beginning and end of the plot.  See the attached screenshot.  Do these also affect other browsers etc?

[image]

neno-k, 2019-03-03:
hi @edwinhayes, yup we did push an update on friday to resolve the issue with safari, in terms of your screenshots, yes this is how the graphs look in other browsers as well, at the end of the plot data is still coming in, so depending on the scale of the graph the line will usually be lower and go up as the graph is refreshed, we will investigate the drop in the beginning

rocos-david, 2019-04-02:
Bugs are resolved.  The drop off at the end of the graph is related to the amount of data received for the final time frame so this will remain.

closed 2019-04-02 by rocos-david