team-rocos/platform #24: What does the 'sample interval' in a storage stream determine?

edwinhayes, 2019-02-28:


neno-k, 2019-03-03:
This is the frequency at which the agent will send data to the backend for a particular telemetry. The value is in ms, so for example if you put 5000 for /mavlink/vfrhud, every 5 seconds the vfrhud data will be sent to the server

closed 2019-03-03 by neno-k

edwinhayes, 2019-03-03:
@neno-k - but, per your documentation, the key difference between a telemetry stream and a storage stream is that while a telemetry stream provides real-time data at some rate, a storage stream transfers _all_ the data of that type, though not in real-time fashion.

So I'd expect to need to specify the sample interval for telemetry widgets, but not for storage streams.  However at present the inverse is true.

neno-k, 2019-03-03:
Hi @edwinhayes , when one opens a teleops connection the data is streamed from the agent at the highest possible frequency in real-time. With the Storage Data Streams we let the user manage what telemetry data the agent sends and at what frequency depending on their needs - I have made an action point on this to review the documentation and make this more clear

rocos-david, 2019-03-03:
@neno-k suggest opening a ticket feature request for 'sample interval' for teleop streams.  If a sensor is sending out 100Hz by default we may need to let the user tune it down.  @edwinhayes you have correctly pointed out that for data streams it will often be 'all' the data, UX has an update to make it clear how to capture all (vs setting it to 0).

edwinhayes, 2019-03-03:
@neno-k  Well to be honest, I think you should probably give further consideration to whether this design makes sense:

Broadly, I would expect that it's the telemetry streams which one would potentially want to decimate, because that allows managing the the real-time bandwidth use, which might have an impact on latency.

Where as generally (if you leave aside the issue of consuming up long term storage resources), I'd want the storage stream to gather _all_ the available data to allow later analysis at a higher resolution.

edwinhayes, 2019-03-03:
@rocos-david - Yep, pretty much that.