team-rocos/platform #26: Agent stops working after some time.

edwinhayes, 2019-03-07:
After running for [not sure exactly, many days], the Agent has lost it's connection to the cloud, citing an authentication issue:

`level=error msg="live stream: received graph error rpc error: code = Unauthenticated desc = invalid auth token: Token is expired:"`

Restarting the agent restores normal operation.

edwinhayes, 2019-03-07:
To clarify, this is an agent on a real robot.

rocos-david, 2019-03-08:
Thanks, known issue with this version of the agent, doesn't auto update token on expiration.  The next major release will solve this along with several other enhancements.

rstinear, 2019-04-08:
Fixed in release [v0.0.5-3163](https://docs.rocos.io/beta/docs/agent-release-notes)

closed 2019-04-08 by rstinear