team-rocos/platform #27: When do changes to stream definitions get applied?

edwinhayes, 2019-03-08:
So, I turned off a stream, yet a couple of minutes later your graphs still seem to show data being transferred.  Is this expected behaviour?

[image]

rocos-david, 2019-03-08:
Will review.  Those graphs do not currently update automatically, you need to click the refresh button. On the roadmap is the graph auto updating every 5 seconds if you are not zoomed in. Can you confirm if this is still not working if you click on refresh?

edwinhayes, 2019-03-08:
Yep, I've smashed refresh a bunch of times.

edwinhayes, 2019-03-08:
Further information, I came across this issue because I added a new piece of data to an existing stream definition, but founded it didn't immediately start coming through.  So if I had to guess, I guess the issue affects modifying stream definitions, not just enabling/disabling them?

rocos-david, 2019-03-08:
Can replicate, backend team will resolve.

rocos-david, 2019-04-02:
Bug fix related to streaming updates released.

closed 2019-04-02 by rocos-david