team-rocos/platform #30: Instantaneous Traffic Measurement

edwinhayes, 2019-03-12:
How can I see (either as a text scalar or a graph), the instantaneous total bandwidth coming off a single robot?  I note that I can see traffic usage related to storage streams, but not telemetry, and because multiple people might be logged into the portal and looking at a robot at once, I can't tell how much bandwidth is being used for telemetry in total.

This might be important if I'm actually flying a robot, and some engineer decides that now would be a good moment to look at a bunch of graphs, and I need to be able to shout at them to stop using up the telemetry bandwidth!

rocos-david, 2019-03-12:
Currently no easy way to see this.  I will flag as a feature request.  In Data Explorer there is the system/network array which provides total Rx and Tx but it's not practical to use for your purpose.  There will be a widget shortly to put that info on the Dashboard but ultimately what you need is the same set of graphs as Robot / Data Stream but for the Teleoperation data.

edwinhayes, 2019-03-12:
I guess the more general feature is that, if you're going to allow teleoperation in a platform which supports multiple users, there will need to be some controls in place to ensure insanity doesn't ensue.