team-rocos/platform #31: Control of Y axis on graphs.

edwinhayes, 2019-03-12:
It would be nice to be able to control the scale of the Y axis in graphs a bit more to be able to force them to display a particular range (usually zero), so that rather than getting a really zigzag line which shows tiny fluctuations in value, you can get a nice flat line which shows that everything is normal.

For instance, you don't really care about your free memory being 2.0000GB or 2.0001GB, you just want to be able to glance at the screen and see a flat line that's well above zero!

rocos-david, 2019-03-12:
For sure, on the roadmap.

rocos-david, 2019-04-02:
This is deployed now in v0.1.0-3042.  https://app.developerhub.io/docs/beta/docs/portal-releases-notes

closed 2019-04-02 by rocos-david