team-rocos/platform #36: Agent init missing config file.

edwinhayes, 2019-09-09:
**Describe the bug**
Trying to run `sudo rocos-agent --init` reports:
`FATA[0000] config file could not be read: Config File "agent" Not Found in "[/etc/opt/rocos-agent /THE/PATH/TO/MY/PWD]" component=config`

There is no such file at `/etc/opt/rocos-agent/agent`, but even if I touch one there, nothing changes.  Also, I note that when attempting to run the init automatically (which will mean as root), the resulting logs are slightly different: ...`Not Found in "[/ /THE/PATH/TO/MY/PWD]"`

**To Reproduce**
1. Clean install of agent r5180.
2. `sudo rocos-agent --init``

**Expected behavior**
Init runs.

**Computer:**
 - OS: Ubuntu 18.04 on x64.

rstinear, 2019-09-09:
Hi @edwinhayes, can you please let me know what it is in your /etc/opt/rocos-agent directory? So sudo ls /etc/opt/rocos-agent

edwinhayes, 2019-09-09:
Ah, sorry, I may have screwed up the chain of evidence there.  I think from memory there was a single file named `agent.json`?  But then I think I deleted it accidentally while I was attempting to see whether a file named `agent` or no-file-at-all made any difference (it didn't as far as I could tell).

edwinhayes, 2019-09-09:
At this time, the directory is empty; running the agent doesn't seem to populate it with anything.

edwinhayes, 2019-09-09:
I'll uninstall the package and give it another go.

rstinear, 2019-09-09:
Cool, yeah - let us know how a re-install goes.

edwinhayes, 2019-09-09:
Ok yeah: so I get the same error, but the directory does contain `agent.json` (which seems to contain a legit looking configuration).  So the issue might be that it's looking in the wrong path?  My PWD during all this is a directory totally unrelated to the agent.

rstinear, 2019-09-09:
Thanks. It's an odd one that we're struggling to reproduce, can you please apt-get remove it then re-install it with sudo apt-get install, and send me the full output from the apt-get install step?

edwinhayes, 2019-09-09:
Any help?  This is with `--print-uris` - is there any other more verbose option?

[tmp.txt](https://github.com/team-rocos/platform/files/3588706/tmp.txt)

edwinhayes, 2019-09-09:
Aaahhh, hang on.  Now it's working?!

edwinhayes, 2019-09-09:
Ok, I understand now; I'm running the wrong executable.  You create two executables with the same name: `/usr/bin/rocos-agent` and `/opt/rocos-agent/bin/rocos-agent`.  The former turns out to be a shell script which calls the latter, but sets some important environment first.  Calling the latter directly is what fails, just in a non-obvious manner.  I don't think this was always the case.

rstinear, 2019-09-09:
Ah good find. Were you in the bin folder when you were running rocos-agent, or some other directory? We need to disambiguate the two executables obviously...

edwinhayes, 2019-09-09:
No, I was just in some totally unrelated folder, but I was running using (what I thought was) the absolute path, not just `rocos-agent` (which would have worked, since it's the correct executable which is on PATH).  This is since normally I'm working via ansible and systemd, which want absolute paths.  Yeah, if one executable was named differently, or if the error was a bit more verbose, this wouldn't have happened.

rstinear, 2019-09-09:
Okay I'll close this, we've added a backlog item to resolve this.

closed 2019-09-09 by rstinear