team-rocos/platform #39: MAVLink demos working badly?

edwinhayes, 2019-09-19:
The creation of virtual MAVLink demo aircraft seems partly broken at the moment?

Applies to both creating a virtual robot in an existing project, and to the two virtual robots created by default in a new MAVLink example project.  Confirmed in both Safari and Chrome with a clean cache.

* Deploying seems to take an unreasonably long time (like twenty minutes).
* Some of the resulting virtual dashboards don't seem to publish all the messages expected; half the default widgets on the dashboard page indicate no-data, but _which_ widgets is not always the same.
* There aren't any canonical mappings, so no global position.

edwinhayes, 2019-09-19:
@rstinear - Can you confirm whether this is a known thing or just me? Would like to be able to give Rob & Linda a demo, following up from our meeting yesterday!

rstinear, 2019-09-19:
Hi Edwin, we're currently experiencing a production issue with one of the underlying nodes the virtual robots are deployed to. As a workaround for now I suggest deploying new virtual craft which will be created on a healthy node. We'll let you know when all is back to normal.

closed 2019-09-22 by rstinear